janw-cz/JWA_BME280 #1: Code won't compile in Arduino Studio 1.6.11

BomberX, 2016-09-19:
Hello

Nice work !
To make your code compile under Arduino 1.6.11, please remove static keyword in the implementation of your static member functions in the .cpp file. In C++, you only have to declare static member function in the .h files.
With this modification, your code compiles successfully in the latest arduino compiler without modifying any other options. I managed to have it compiled with an ESP8266 target (ESP-12E module).
I could not test it since I am missing the BME280 board, but I will as soon as I can get one.

vodnik12, 2016-09-20:
Hi,
thank you for reporting that. 

I have removed the static keyword from all function definitions in the .cpp file. Please let me know whether the library works for you when you get the sensor.

Jan

closed 2016-09-20 by vodnik12

BomberX, 2016-10-03:
Hello Jan 

As promised, just for your information: I received my sensor today (this one https://www.amazon.fr/Capteur-Pression-Barom%C3%A9trique-Dhumidit%C3%A9-Temp%C3%A9rature/dp/B01GQ3T1A4/keywords=STK0151002635 ), and it's working fine with an ESP8266 board (this one : https://www.amazon.fr/gp/product/B01GCK3J40/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1 ). 

Thanks again for sharing your work ! 

Fabien 

----- Mail original -----