janw-cz/JWA_BME280 #2: Compile errors using Arduino IDE 1.8.3

lloydch, 2017-08-07:
I have been using JWA_BME280 version 0.8.0 since last December and have been very happy with it.    Over the past couple of days I deleted the IDE and libraries because I had a mess and then installed the IDE again and started installing the latest versions of the libraries.       I noticed changes in the new 0.9.0 and I have corrected those in my sketches but I am still getting compile errors even with your example sketch.

`Arduino: 1.8.3 (Linux), Board: "Arduino Nano, ATmega328"

/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp: In member function 'bool Bme280BoschWrapper::beginI2C(uint8_t)':
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp:21:10: error: 'struct bme280_dev' has no member named 'id'
   bme280.id = dev_addr;
          ^
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::I2CInit()':
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp:91:10: error: 'struct bme280_dev' has no member named 'interface'
   bme280.interface = BME280_I2C_INTF;
          ^
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::SPIInit()':
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp:101:10: error: 'struct bme280_dev' has no member named 'id'
   bme280.id = 0;
          ^
/home/neddy/Arduino/libraries/JWA_BME280-master/Bme280BoschWrapper.cpp:102:10: error: 'struct bme280_dev' has no member named 'interface'
   bme280.interface = BME280_SPI_INTF;
          ^
exit status 1
Error compiling for board Arduino Nano.
`
Would you mind having a look and checking to see if you get the same error.
Thanks, Lloyd.

vodnik12, 2017-08-08:
Thank you for reporting that. I was probably too fast when doing the upgrade. These bugs are fixed now, you can download the fixed version.

lloydch, 2017-08-10:
Thank you for your quick response and all is working now, closing this issue.
Thanks, Lloyd.

closed 2017-08-10 by lloydch