janw-cz/JWA_BME280 #8: error with ide 1.89

stef-ladefense, 2019-08-03:
In file included from C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp:5:0:

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Wire/Wire.h: In static member function 'static int8_t Bme280BoschWrapper::I2CRead(uint8_t, uint8_t, uint8_t*, uint16_t)':

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Wire/Wire.h:70:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int, int);

             ^

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Wire/Wire.h:68:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t, uint8_t);

for information

janw-cz, 2019-10-09:
Works OK with Arduino 1.8.10. Can you be more specific about what version of Arduino IDE and which board are you using?

stef-ladefense, 2019-11-22:
have same error with 1.8.10 and esp8266 2.61

C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp: In static member function 'static int8_t Bme280BoschWrapper::I2CRead(uint8_t, uint8_t, uint8_t*, uint16_t)':

C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp:122:33: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]

   Wire.requestFrom(dev_addr, cnt);

                                 ^

In file included from C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp:5:0:

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\Wire/Wire.h:70:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int, int);

             ^

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\Wire/Wire.h:68:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t, uint8_t);

janw-cz, 2019-12-08:
Bug was ESP266 specific. Fixed.

closed 2019-12-08 by janw-cz