janw-cz/JWA_BME280 #11: Force mode

Eldenroot, 2020-04-10:
Just one more question - how to use force mode with your library? Can you give me some example?

Thank you!

janw-cz, 2020-04-10:
Hi,
the included example file uses forced mode. It is the default mode of operation with my driver. It is defined by a parameter in the constructor:

`Bme280BoschWrapper bme280(true);` 
true = do measurements in force mode
false = do measurements in continuous mode

Hope that helps,
Jan

Eldenroot, 2020-04-10:
OK, asi oba dva jsme Seci, tedy predpokladam to dle jmena a emailu.

Kdyz bych chtel pouzit neco takoveho:

/*
    // weather monitoring
    Serial.println("-- Weather Station Scenario --");
    Serial.println("forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,");
    Serial.println("filter off");
    bme.setSampling(Adafruit_BME280::MODE_FORCED,
                    Adafruit_BME280::SAMPLING_X1, // temperature
                    Adafruit_BME280::SAMPLING_X1, // pressure
                    Adafruit_BME280::SAMPLING_X1, // humidity
                    Adafruit_BME280::FILTER_OFF   );
                      
    // suggested rate is 1/60Hz (1m)
    delayTime = 60000; // in milliseconds
*/

Dale pri I2C komunikaci to je s touto knihovnou jak? Protoze omlouvam se, ale u Adafruit BME280 knihovny je to asi trosku jinak a bral jsem od tama example... bohuzel se mi s Adafruit knihovnou stane, ze po urcite dobe (treba tyden) kdy bezi v kuse arduino mi BME280 sensor sem tam hodi nahodne hodnotu NAN u teploty... vlhkost a tlak to meri v poradku porad, pri dalsim force mereni (merim kazdych 5 minut) mi to zase zmeri dobre a pak treba zase pul dne pokoj... jakmile ale vypnu a zapnu arduino tak to zase bezi nekolik dni bez problemu. Myslim si, ze je to knihovnou a chcu tedy nnyi zkusi tu tvou jen trosku tapu. Diky

Eldenroot, 2020-04-10:
EDIT: OK uz to vidim v tom prikladu na I2C komunikaci, tudiz muj dotaz jen zni aktualne k tomu sampling, filter atd. Diky

closed 2020-04-10 by Eldenroot