janw-cz/JWA_BME280 #14: Bosch driver 3.5.0 update?

stef-ladefense, 2020-08-01:
https://github.com/BoschSensortec/BME280_driver/releases

merci

janw-cz, 2021-01-23:
Upgraded

stef-ladefense, 2021-01-23:
oups...

C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::I2CInit()':
C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp:106:19: error: invalid conversion from 'void (*)(uint32_t) {aka void (*)(unsigned int)}' to 'bme280_delay_us_fptr_t {aka void (*)(unsigned int, void*)}' [-fpermissive]
   bme280.delay_us = Bme280BoschWrapper::delaymsec;
                   ^
C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::SPIInit()':
C:\Users\stephane\Documents\Arduino\libraries\JWA_BME280-master\Bme280BoschWrapper.cpp:116:19: error: invalid conversion from 'void (*)(uint32_t) {aka void (*)(unsigned int)}' to 'bme280_delay_us_fptr_t {aka void (*)(unsigned int, void*)}' [-fpermissive]
   bme280.delay_us = Bme280BoschWrapper::delaymsec;

stef-ladefense, 2021-01-23:
i found folk in ver 3.5 working
https://github.com/h2zero/JWA_BME280

janw-cz, 2021-01-23:
Should be better now. The error did not show up in my Arduino, weird.

stef-ladefense, 2021-01-23:
it's work now for me
thk Jan

closed 2021-01-31 by janw-cz