janw-cz/JWA_BME280 #15: Compile error runnin latest JWA_BME280 on Arduino IoT

wandrols, 2021-01-31:
Hi,

I am writing a Sketch using MKR1010 Wifi, a BMP280 humidity sensor and Arduino IoT Cloud + web editor. A few days ago when compiling I got below compile error including Bme280BoschWrapper. It had work worked before.

/home/builder/opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware /home/builder/opt/arduino-builder/hardware -hardware /home/builder/.arduino15/packages -tools /home/builder/opt/arduino-builder/tools -tools /home/builder/.arduino15/packages -built-in-libraries /home/builder/opt/libraries/latest -logger humantags -fqbn arduino:samd:mkrwifi1010 -build-cache /tmp -build-path /tmp/158271934/build -verbose -prefs runtime.tools.CMSIS.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs runtime.tools.CMSIS-Atmel.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs runtime.tools.arduinoOTA.path=/home/builder/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs runtime.tools.arm-none-eabi-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs runtime.tools.bossac.path=/home/builder/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs runtime.tools.openocd.path=/home/builder/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs runtime.tools.linuxuploader.path=/home/builder/.arduino15/packages/arduino/tools/linuxuploader/1.5.1 -prefs runtime.tools.arduino101load.path=/home/builder/.arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs runtime.tools.flashpack.path=/home/builder/.arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs runtime.tools.i586-poky-linux-uclibc.path=/home/builder/.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0 -prefs runtime.tools.sketchUploader.path=/home/builder/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0 -prefs runtime.tools.avr-gcc.path=/home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs runtime.tools.mkspiffs.path=/home/builder/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs runtime.tools.core2-32-poky-linux.path=/home/builder/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0 -prefs runtime.tools.dfu-util.path=/home/builder/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino2 -prefs runtime.tools.ctags.path=/home/builder/.arduino15/packages/builtin/tools/ctags/5.8-arduino11 -prefs runtime.tools.serial-discovery.path=/home/builder/.arduino15/packages/builtin/tools/serial-discovery/1.1.0 -prefs runtime.tools.esptool.path=/home/builder/.arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9 -prefs runtime.tools.avrdude.path=/home/builder/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs runtime.tools.x86-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/x86-linux-gcc/7.2.0 -prefs runtime.tools.arc-elf32.path=/home/builder/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs runtime.tools.arm-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-linux-gcc/4.9.3 -prefs runtime.tools.nrf5x-cl-tools.path=/home/builder/.arduino15/packages/arduino/tools/nrf5x-cl-tools/9.3.1 -prefs runtime.tools.xtensa-lx106-elf-gcc.path=/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -libraries /tmp/158271934/custom -libraries /tmp/158271934/pinned /tmp/158271934/Embla_jan09a

Using library SNU at version 1.0.2 in folder: /home/builder/.arduino15/packages/arduino/hardware/samd/1.8.9/libraries/SNU

Using library SPI at version 1.0 in folder: /home/builder/.arduino15/packages/arduino/hardware/samd/1.8.9/libraries/SPI

/home/builder/opt/libraries/latest/jwa_bme280_1_0_3/Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::I2CInit()':

/home/builder/opt/libraries/latest/jwa_bme280_1_0_3/Bme280BoschWrapper.cpp:106:41: error: invalid conversion from 'void (*)(uint32_t) {aka void (*)(long unsigned int)}' to 'bme280_delay_us_fptr_t {aka void (*)(long unsigned int, void*)}' [-fpermissive]

bme280.delay_us = Bme280BoschWrapper::delaymsec;

^~~~~~~~~

/home/builder/opt/libraries/latest/jwa_bme280_1_0_3/Bme280BoschWrapper.cpp: In member function 'void Bme280BoschWrapper::SPIInit()':

/home/builder/opt/libraries/latest/jwa_bme280_1_0_3/Bme280BoschWrapper.cpp:116:41: error: invalid conversion from 'void (*)(uint32_t) {aka void (*)(long unsigned int)}' to 'bme280_delay_us_fptr_t {aka void (*)(long unsigned int, void*)}' [-fpermissive]

bme280.delay_us = Bme280BoschWrapper::delaymsec;

^~~~~~~~~

exit status 1

janw-cz, 2021-01-31:
Thank you for reporting that. I made a mistake when marking new version to be pulled into Arduino libraries. Version 1.0.4 has that bug already fixed, but Arduino IDE has only version 1.0.3. We will have to wait until Library manager refreshes the version to 1.0.4.

wandrols, 2021-01-31:
Hi and thanks for info. Do you know with which frequency Library Manager refreshes? I think the issue has been there for about 1 week now?

janw-cz, 2021-01-31:
It should be once a day. In the Library manager, you can select previous version of the library, which works. Until v1.0.4 becomes available through standard channel.

wandrols, 2021-01-31:
Ok thanks. I can select specific versions in Library manager in the Arduino desktop client, but seem not to be able to do that in the web editor Library manager, it is by default the latest. So to compile for IoT Cloud in web editor I guess we will have to wait until v1.0.4 becomes available then.

janw-cz, 2021-01-31:
Arduino IDE now shows the latest version.

closed 2021-09-11 by janw-cz